RoboticsClubatUCF/Bowser #20: Research alternative mapping packages

chickenfromouterspace, 2019-10-02:
Document a comparison of multiple mapping packages. Determine which mapping package best suits the needs of IGVC.

chickenfromouterspace, 2019-10-09:
Look into rtabmap.

BriandBlalock, 2019-10-15:
I will also be looking into this as part of my senior design

BriandBlalock, 2019-10-17:
Rtab-map info 
Ros wiki: http://wiki.ros.org/rtabmap_ros
Github/website: http://introlab.github.io/rtabmap/

BriandBlalock, 2019-10-20:
THe RTAB full paper. It also does a comparison of multiple other mapping methods.  

https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/7/7a/Labbe18JFR_preprint.pdf

BriandBlalock, 2019-10-23:
➤ Robotics Club at UCF commented:

We are having some preliminary success with RTAB-Map and the ZED camera. We have successfully created a (limited) 3D occupancy grid

closed 2020-01-29 by BriandBlalock