RoboticsClubatUCF/Bowser #24: Find our how to isolate the ground plane from the Lidar data

BriandBlalock, 2019-10-15:
An useful feature for the navigation code would be the ability to isolate just the ground data from the Lidar data. A possible approach would be to get the xyz point data and isolate anything between a specific Z.

chickenfromouterspace, 2019-10-15:
To start , go into the detect_lidar.py at Bowser/catkin_ws/src/bowser_lidar/nodes/detect_lidar.py and figure out how to isolate the z variable.

self.points is a 2D array where the first dimension represents all the points scanned and the second dimension is x, y, and z. It should be possible to isolate z with a for-loop that accesses self.points[i][2].

BriandBlalock, 2019-10-16:
If possible, try to use numpy functions rather than manually looping through the data. Numpy is over 20x faster