RoboticsClubatUCF/Bowser #25: Figure out the catkin_make error

BriandBlalock, 2019-10-16:
There is currently an issue when you try to run catkin_make. It will give you an error relating to CmakeLists.txt , this may be due to it being pushed to the repository when it shouldn't be. the .gitignore may need to be fixed

chickenfromouterspace, 2019-10-16:
Why did we decide to place the entire catkin workspace in the repo instead of all the individual packages we use? It should be possible to git clone a folder with all our packages into the src folder without issue.

Put simply, Bowser/catkin_ws/src causes more issues than catkin_ws/src/Bowser.

If there is a good reason though, we would probably need to a procedure for [workspace overlaying](http://wiki.ros.org/catkin/Tutorials/workspace_overlaying) to keep things stable.

BriandBlalock, 2019-10-17:
We did it to make it easier for new people to just clone the repo and start working

chickenfromouterspace, 2019-10-22:
Are there any screenshots to show what the error specifically is?

closed 2020-01-29 by BriandBlalock